NeuroTechX/EEG-ExPy #233: is it possible to save the PPG data from Muse 2 (with bluemuse and/or brainflow)?

danielemarinazzo, 2023-05-09:
## 📝 Provide a description of the new feature

Hi, unless I have missed something, is not possible to save the PPG file from the Muse 2, right? I can start the stream with bluemuse but then the PPG is not saved.
Is this something I can add myself, or that can be added with not too much effort?
It can be also with brainflow.

JohnGriffiths, 2023-05-10:
Ciao @danielemarinazzo 

We haven't tried using the ppg and accelerometer data in eegnb. Mainly because it hasn't been straightforward to access that data previously. 

Blusemuse does send the ppg chans, to a separate lsl stream as the eeg. You can see this in the blue muse gui. 

The preferred streaming option is now Brainflow not Blusemuse however.

Previously Brainflow did not support anything other than the eeg chans from muse. However I just took a look at the [docs](https://brainflow.readthedocs.io/en/stable/SupportedBoards.html#muse) and it looks like it does indeed now allow accelerometer and ppg chans to be pulled also - 

```
Available BrainFlow Presets:

BrainFlowPresets.DEFAULT_PRESET, it contains EEG data, to enable 5th EEG channel use board.config_board("p50")

BrainFlowPresets.AUXILIARY_PRESET, it contains Gyro and Accel data, enabled by default

BrainFlowPresets.ANCILLARY_PRESET, it contains PPG data, to enable it use board.config_board("p61")

```

I think it would be a very good idea to switch to always grabbing those additional streams by default in the eegnb EEG muse device class,  and output cav file, IF doing so does not prove too difficult.

One difficulty that could arise is that the sampling rate is  different  for the ppg to the eeg. 

A PR attempting this would be superb! Remember to fork from `develop` when doing so, not `main`. 

Thoughts?

tmorshed, 2023-10-05:
So, one idea is to add a new device type to the `devices.eeg` board type, e.g., named "museX_bfp":
https://github.com/NeuroTechX/eeg-notebooks/blob/ac1f5fb14d3c2baa8be0cc6ef0cfb87671060e95/eegnb/devices/eeg.py#L33

This could use the pertinent board config from brainflow and get the new channels. Updating the old device types may be tricky as it may add dimensions and mess up with legacy code and experiments/analyses.

Also see: https://github.com/brainflow-dev/brainflow/blob/master/python_package/examples/tests/muse_ppg.py

khuzaymahbinharis-jpg, 2026-08-19:
﻿I'd like to take this on as a first contribution.

**Approach:** keep the existing EEG CSV unchanged (same columns and sampling rate) so analysis/experiments stay compatible, and write Muse PPG and accel/gyro as sidecar files:

- `recording_....csv` — EEG + stim (current format)
- `recording_...._ppg.csv` — PPG from `ANCILLARY_PRESET`
- `recording_...._accel.csv` — accel/gyro from `AUXILIARY_PRESET`

This follows BrainFlow's split-preset model (EEG ~256 Hz, accel ~52 Hz, PPG ~64 Hz) instead of concatenating streams into one table.

**Device scope:** BrainFlow Muse 2 / Muse S (`muse2_bfn`, `muse2_bfb`, `museS_bfn`, `museS_bfb`). Enable PPG with `p51` (Muse 2) / `p61` (Muse S) so we get PPG without adding a 5th EEG channel. BlueMuse/`muselsl` and Muse 2016 are out of scope for this PR.

**Base branch:** I'll target `master` rather than `develop`, since `master` is the current default and recent PRs land there.

I don't have a Muse headset for hardware validation; the PR will include mocked tests and ask for a maintainer with a device to confirm.

Happy to adjust (e.g. a separate `museX_bfp` device name) if you'd rather not enable the extra streams by default.