NeuroTechX/EEG-ExPy #236: Problems with saving data file using brainflow

larastellamarie, 2023-05-20:
Hello,

I am currently trying to record data with the Muse 2 using the SSAEP experiment and a MMN experiment. When I run them apart using brainflow everything goes fine. Now I tried running the two experiements shortly after each other which results in only one data file being saved. The other experiment runs but does not save a data file in the end and produces this error:

[2023-05-17 22:13:08.678] [board_logger] [error] failed to send command h to device
Traceback (most recent call last):
  File "C:\Users\Asus\anaconda3\envs\eeg-notebooks\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Users\Asus\anaconda3\envs\eeg-notebooks\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Asus\anaconda3\envs\eeg-notebooks\Scripts\eegnb.exe\__main__.py", line 7, in <module>
  File "C:\Users\Asus\anaconda3\envs\eeg-notebooks\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Asus\anaconda3\envs\eeg-notebooks\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\Asus\anaconda3\envs\eeg-notebooks\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Asus\anaconda3\envs\eeg-notebooks\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Asus\anaconda3\envs\eeg-notebooks\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Asus\eeg-notebooks\eegnb\cli\__main__.py", line 90, in runexp
    run_experiment(experiment, eeg, recdur, outfname)
  File "C:\Users\Asus\eeg-notebooks\eegnb\cli\utils.py", line 56, in run_experiment
    module.present(duration=record_duration, eeg=eeg_device, save_fn=save_fn)  # type: ignore
  File "C:\Users\Asus\eeg-notebooks\eegnb\experiments\auditory_ssaep\ssaep_onefreq.py", line 117, in present
    eeg.stop()
  File "C:\Users\Asus\eeg-notebooks\eegnb\devices\eeg.py", line 429, in stop
    self._stop_brainflow()
  File "C:\Users\Asus\eeg-notebooks\eegnb\devices\eeg.py", line 317, in _stop_brainflow
    self.board.stop_stream()
  File "C:\Users\Asus\anaconda3\envs\eeg-notebooks\lib\site-packages\brainflow\board_shim.py", line 1211, in stop_stream
    raise BrainFlowError('unable to stop streaming session', res)
brainflow.exit_codes.BrainFlowError: BOARD_WRITE_ERROR:4 unable to stop streaming session

If I use bluemuse everything works fine. I don´t think that it has anything to do with the order (sometimes the first experiment elicits an error and sometimes the second) or the experiment itself. Any advice how I can avoid this problem?

oreHGA, 2023-11-16:
Hey, @larastellamarie , is this error still occurring for you?

larastellamarie, 2023-11-16:
> Hey, @larastellamarie , is this error still occurring for you?

Hey,

I don´t know, I am sorry. I switched to using Bluemuse after brainflow didn´t work for me but I am encountering new problems for which I opened a new issue