NeuroTechX/EEG-ExPy #247: Incomplete outputfile SSAEP_onefreq with Bluemuse

larastellamarie, 2023-11-16:
## ℹ Computer information

- Platform OS: Windows
- Brain Interface Used: Muse

Hello,

I have a new issue with the output files of the SSAEP_onefreq script. I am using bluemuse to record sessions of 15 minutes. The problem is that sometimes the outputfile is almost empty or very short and does not contain the full 15 minutes of recorded data. 
I am assuming that it has something to do with the stream transmission breaking off but there are no error messages printed. 
The experiment also runs the requested 15 minutes, the problem is just the output file where the data is missing in the end. Is anybody else having this problem? Does anybody have advice on how to fix this?

Thank you in advance for your help!

JohnGriffiths, 2024-06-06:
Hi lara. 

I know it's been a while but just picking up on this Q. 

Is this a recurrent issue and are you still having it?

In general we recommend using brainflow rather than bluemuse backend for streaming; is more stable and universal. 

If the signal drops during a recording there will be some message on this in the terminal window. But it wouldn't necessarily yield an error message.

larastellamarie, 2024-09-10:
Hi,

Thank you for your reply! I did not run the experiment for a while now but the issue appeared quite often when me and my colleague were collecting data. Of the 60 sessions we completed we had around 15 incomplete sessions even though there was no message in the window that the stream was dropped. I tried using brainflow but it didn´t work on my computer.