NeuroTechX/EEG-ExPy #277: syntax error in muselsl

danielemarinazzo, 2024-10-10:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc):  MAC M2
- Python Version: 3.7
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc): Muse

## 📝 Provide detailed reproduction steps (if any)

We installed everything on a Mac M2 OS 14.3.1 (23D60)

### ✔️ Expected result

running eegnb -ip and go to the next step

### ❌ Actual result

when running eegnb runexp -ip (and when running the test cell as in the installation guide) we get this syntax error message (corresponding to the colon after "device")

  File "/Users/USERNAME/miniconda3/envs/eeg-notebooks/lib/python3.7/site-packages/muselsl/muse.py", line 90
    if ((device := self.adapter.connect(self.address, retries)) 

Duplicate from https://github.com/alexandrebarachant/muse-lsl/issues/216#issue-2578472803

JohnGriffiths, 2024-10-10:
hi Daniele. 

1 - can you share a screenshot of the exact terminal commands and output

2 - recommended device type to use is `muse2_bfn`, which uses brainflow not muselsl as backend.

danielemarinazzo, 2024-10-10:
[image]

How to make muse2_bfn default? A previous error complained about brainflow not being installed

BoBo262, 2024-10-18:
Ran Into the same problem. Tried to run the installation test but got a SyntaxError related to muselsl
[image]

ErikBjare, 2024-10-18:
That syntax (the "walrus operator") is only supported in Python 3.8+

BoBo262, 2024-10-18:
Thanks for the advice. I tried to create a new environment for the eeg notebook, this time in 3.8 but got a new error 
[image]

oreHGA, 2024-10-21:
@BoBo262 , can you try the following and let us know what you get:

- creating the eeg-expy environment without wxpython
`conda create -n "eeg-expy" python=3.8 git pip`

- change directory to where eeg-expy is downloaded on your machine and then install
`cd <your_path_where_eegexpy_is_on_your_machine>`
`pip install -e .`