NeuroTechX/EEG-ExPy #289: experiment does not start

danielemarinazzo, 2024-12-20:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc): Windows
- Python Version: 3.8
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc): Muse 2

## 📝 Provide detailed reproduction steps (if any)

Installation ok (had to manually install rust otherwise install did not work; then click, psychopy, muselsl, otherwise the select interface would not start).

selected muse 2 brainflow native

### ✔️ Expected result

can proceed to select experiment and start

### ❌ Actual result

nothing happens

## 📷 Screenshots
[image]

_Are there any useful screenshots? WinKey+Shift+S (Windows) / Shift+Command+4 (MacOS) and then just paste them directly into the form_

pellet, 2025-03-17:
Hi @danielemarinazzo, I don't have a muse 2 handy unfortunately, only have an openbci cyton for testing.  It looks like it is aborting early after selecting option 6, before making it to the experiment selection stage.  
Due to there being no error or exceptions logged it makes it hard to diagnose the exact issue.  Have you tried using option 2 or option 7 to see if they work or give more useful information?

danielemarinazzo, 2025-03-17:
They also didn't work. It should be an incompatibility with some newer version, possibly of psychopy. It was a time sensitive issue so I aborted the use on that pc where I had just performed a new installation and reverted to a pc with an older installation that worked.

pellet, 2025-03-17:
ok thanks for letting me know, I'll keep an eye out for any changes which could be causing a regression.